formatBCE/Respeaker-Lite-ESPHome-integration #1: Component not found: Audio

jamesgid-gc, 2024-10-17:
Hi formatBCE,

I really appreciate your esphome-respeaker-lite.yaml code. It has formed the basis for my testing of of the new voice-kit code from esphome on my ReSpeaker Lite kit. Overnight, I am now getting an error of 'Component not found: audio' when editing the config. The error is highlighted at line 1. It is odd because there is no reference to an audio component, only i2s_audio. I did update esphome yesterday, so I think it might have something to do with that. I tried the beta and dev channels of esphome also but the problem remains.

Are you seeing this also?

Regards,

formatBCE, 2024-10-18:
There was a breaking change today. I'll update YAML in few hours.

formatBCE, 2024-10-18:
Sorry, driving and can't write much. It's update in Nabu component. It actually brings a lot of stability and performance to media player.

There's update to XMOS firmware today too. If you have experience in writing ESPHome custom components (i2c), you could help a lot to bring actual programmatic "mute" state to the project.

formatBCE, 2024-10-18:
@jamesgid-gc updated XMOS firmware and YAML. Please update. :)

closed 2024-10-18 by formatBCE

jamesgid-gc, 2024-10-19:
You are a legend. I will try and test it this afternoon. Cheers.