formatBCE/Respeaker-Lite-ESPHome-integration #23: Disable Mute Mic in yaml

camcoding, 2024-11-20:
First off, thank you so much for your efforts on this Respeaker Lite project!  

I have the respeaker lite voice kit from Seed and went through your install notes using the  [esphome-respeaker-lite.yaml] file.  However, I currently don't have any soldering equipment to perform things "solder USR to D2 and Mute do D3".  I don't understand what USR means and I'm pretty sure Mute is the capability to mute the mic programmatically.  

Is it possible to not solder anything from this Seeed Studio board and alter your yaml file to eliminate USR and Mute features so I could at least test this as a remote voice assistant?

Here's a snippet of my logs:
INFO ESPHome 2024.10.3
INFO Reading configuration /config/esphome/respeaker.yaml...
INFO Updating https://github.com/esphome/voice-kit@dev
INFO Updating https://github.com/esphome/esphome.git@pull/7605/head
INFO Updating https://github.com/formatBCE/Respeaker-Lite-ESPHome-integration@main
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.86.36 using esphome API
INFO Successfully connected to respeaker-satellite @ 192.168.86.36 in 0.198s
INFO Successful handshake with respeaker-satellite @ 192.168.86.36 in 0.049s
[10:16:05][I][app:100]: ESPHome version 2024.10.3 compiled on Nov 19 2024, 21:35:15
[10:16:05][C][wifi:600]: WiFi:
[10:16:05][C][wifi:428]:   Local MAC: F0:9E:9E:3B:4E:A0
[10:16:05][C][wifi:433]:   SSID: [redacted]
[10:16:05][C][wifi:436]:   IP Address: 192.168.86.36
[10:16:05][C][wifi:440]:   BSSID: [redacted]
[10:16:05][C][wifi:441]:   Hostname: 'respeaker-satellite'
[10:16:05][C][wifi:443]:   Signal strength: -56 dB ▂▄▆█
[10:16:05][C][wifi:447]:   Channel: 11
[10:16:05][C][wifi:448]:   Subnet: 255.255.255.0
[10:16:05][C][wifi:449]:   Gateway: 192.168.86.1
[10:16:05][C][wifi:450]:   DNS1: 192.168.86.1
[10:16:05][C][wifi:451]:   DNS2: 0.0.0.0
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][C][logger:185]: Logger:
[10:16:05][C][logger:186]:   Level: DEBUG
[10:16:05][C][logger:188]:   Log Baud Rate: 115200
[10:16:05][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][C][i2c.idf:075]: I2C Bus:
[10:16:05][C][i2c.idf:076]:   SDA Pin: GPIO5
[10:16:05][C][i2c.idf:077]:   SCL Pin: GPIO6
[10:16:05][C][i2c.idf:078]:   Frequency: 50000 Hz
[10:16:05][C][i2c.idf:084]:   Recovery: bus successfully recovered
[10:16:05][I][i2c.idf:094]: Results from i2c bus scan:
[10:16:05][I][i2c.idf:100]: Found i2c device at address 0x18
[10:16:05][I][i2c.idf:100]: Found i2c device at address 0x42
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][C][switch.gpio:068]: GPIO Switch 'mute_toggle'
[10:16:05][C][switch.gpio:091]:   Restore Mode: always OFF
[10:16:05][C][switch.gpio:031]:   Pin: GPIO4
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][C][esp32_rmt_led_strip:187]: ESP32 RMT LED Strip:
[10:16:05][C][esp32_rmt_led_strip:188]:   Pin: 1
[10:16:05][C][esp32_rmt_led_strip:189]:   Channel: 0
[10:16:05][C][esp32_rmt_led_strip:214]:   RGB Order: GRB
[10:16:05][C][esp32_rmt_led_strip:215]:   Max refresh rate: 0
[10:16:05][C][esp32_rmt_led_strip:216]:   Number of LEDs: 1
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'User button'
[10:16:05][C][gpio.binary_sensor:016]:   Pin: GPIO3
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][W][respeaker_lite:069]: Read mic mute state failed
[10:16:05][C][light:103]: Light 'respeaker-satellite'
[10:16:05][C][light:105]:   Default Transition Length: 0.0s
[... 61 lines cut ...]
[10:16:06][C][safe_mode:020]:   Boot considered successful after 60 seconds
[10:16:06][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:16:06][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[10:16:06][W][respeaker_lite:069]: Read mic mute state failed
[10:16:06][C][api:140]: API Server:
[10:16:06][C][api:141]:   Address: respeaker-satellite.local:6053
[10:16:06][C][api:145]:   Using noise encryption: NO
[10:16:06][W][respeaker_lite:069]: Read mic mute state failed
[10:16:06][C][micro_wake_word:072]: microWakeWord:
[10:16:06][C][micro_wake_word:073]:   models:
[10:16:06][C][micro_wake_word:015]:     - Wake Word: Okay Nabu
[10:16:06][C][micro_wake_word:016]:       Probability cutoff: 0.97
[10:16:06][C][micro_wake_word:017]:       Sliding window size: 5
[10:16:06][C][micro_wake_word:015]:     - Wake Word: Stop
[10:16:06][C][micro_wake_word:016]:       Probability cutoff: 0.50
[10:16:06][C][micro_wake_word:017]:       Sliding window size: 5
[10:16:06][C][micro_wake_word:021]:     - VAD Model
[10:16:06][C][micro_wake_word:022]:       Probability cutoff: 0.50
[10:16:06][C][micro_wake_word:023]:       Sliding window size: 5
[10:16:06][W][respeaker_lite:069]: Read mic mute state failed
[10:16:06][W][respeaker_lite:069]: Read mic mute state failed
[10:16:06][W][respeaker_lite:069]: Read mic mute state failed
[10:16:06][W][respeaker_lite:069]: Read mic mute state failed
[10:16:06][W][respeaker_lite:069]: Read mic mute state failed
[10:16:06][W][respeaker_lite:069]: Read mic mute state failed

formatBCE, 2024-11-20:
Hey! Thanks for kind words!

Yes, you can delete these:
1. Mute state declaration here: https://github.com/formatBCE/Respeaker-Lite-ESPHome-integration/blob/main/esphome-respeaker-lite.yaml#L143
2. Everything related to mute from `switch:` section: https://github.com/formatBCE/Respeaker-Lite-ESPHome-integration/blob/main/esphome-respeaker-lite.yaml#L154
3. `binary_sensor:` section https://github.com/formatBCE/Respeaker-Lite-ESPHome-integration/blob/main/esphome-respeaker-lite.yaml#L264 (USR is user-defined button o Respeaker board, and it's getting soldered to the GPIO pin same way MUTE does, it all is in Respeaker docs)

I'm pretty sure you'll have to delete https://github.com/formatBCE/Respeaker-Lite-ESPHome-integration/blob/main/esphome-respeaker-lite.yaml#L484 and probably something else (i might overlooked) to get it compiled. But it will work.

camcoding, 2024-11-21:
I'm sure I'm doing something wrong but I'm still getting mute errors after removing all the mute stuff from my yaml file.  Note, I'm able to compile the code into a binary and I see the device's LED flashing and working inside home assistant but I'm seeing a ton of these error messages:
[17:47:37][W][respeaker_lite:069]: Read mic mute state failed
[17:47:37][W][respeaker_lite:069]: Read mic mute state failed
[17:47:37][W][respeaker_lite:069]: Read mic mute state failed
[17:47:37][W][respeaker_lite:069]: Read mic mute state failed

I'll attach my yaml
[my_yaml.txt](https://github.com/user-attachments/files/17838320/my_yaml.txt)

camcoding, 2024-11-21:
I removed all the mic and mute stuff from the yaml, recompiled the binary, and it's still throwing this error:
18:19:56][W][respeaker_lite:069]: Read mic mute state failed.

Attached is my YAML
[my_yaml.txt](https://github.com/user-attachments/files/17839286/my_yaml.txt)

formatBCE, 2024-11-21:
> I removed all the mic and mute stuff from the yaml, recompiled the binary, and it's still throwing this error:
> 18:19:56][W][respeaker_lite:069]: Read mic mute state failed.
> 
> Attached is my YAML
> [my_yaml.txt](https://github.com/user-attachments/files/17839286/my_yaml.txt)
> 

Remove whole Respeaker component declaration. Looks like it is still trying to read mute state...

closed 2024-11-21 by camcoding