formatBCE/Respeaker-Lite-ESPHome-integration #26: Speaker volume/clipping past 70-75%

VACIndustries, 2024-11-30:
Hey @formatBCE !  Great job throwing this together; it really has made the difference nearly making an identical replacement for an Echo or Google Home device.  I threw together the housing too (also awesome) with the same speakers you used (the one the respeaker came with was not great).  It seems like it's better quality but it seems it starts clipping after 70-75% volume.  This is the same that happened with the speaker that the seeedstudio sends with it as well.

I was wondering if you found if there's anyway around this to get more volume without clipping.  It's possible it's just the speaker and something higher quality would do the trick but I'm also not an audio engineer so figured I'd ask if you looked into it.

formatBCE, 2024-11-30:
Hello! :)

Thanks, I'm glad my work is helping people!

Yes, I hear clipping... Looks like it's trouble of Respeaker inbuilt amplifier, because it happens with 5W, 3W speaker, and with headphones in audio jack.

I restricted media player by top volume, it helps a bit.

formatBCE, 2024-11-30:
I actually will try asking Seeed guys.
NabuCasa Voice PE device is using same DAC. Either they have same trouble, or it's XMOS board firmware problem and Seeed devs should look into it.

formatBCE, 2024-12-03:
Ok, i asked Seeed developers, and the answer is that we cannot do anything with speaker. 
Respeaker is using same DAC as PE, but different amplifier. This amp is VBAT=5v, Class AB, which starts showing distortion on around 2.5-3W. So effectively, using 5W speaker on full volume isn't something we want to do...

I'll play with 3W speaker on new enclosure, but it definitely will be quieter. Also i can restrict max volume on amplifier, thus eliminating distortion - but restraining to less volume.

One option to get more juice is to use line-out from Respeaker with some active speaker with good external pre-amp. I guess, it's in the same niche as Echo Dot - small speaker with not-that-good volume, but with line-out in case you want more sound. :)

Closing this so far, as it's hardware-restricted.

closed 2024-12-03 by formatBCE

VACIndustries, 2024-12-03:
Thanks for checking it out!