formatBCE/Respeaker-Lite-ESPHome-integration #27: Enable/Disable Mic Mute via HA Not working

deadphyre, 2024-12-14:
In short, the hardware button to turn on and off the Mic Mute does in dead work, however, the Home Assistant with ESPHome Mic Mute Enable/Disable does not work. While viewing the logs, I first ran the hardware button to disable Mic Mute. I then Click the switch in HA to enable Mic Mute but looks as though it immediately goes to the off state. The small clip of the logs showing this will be attached below. Please let me know if there is more that I can provide.

Using [esphome-respeaker-lite-16kHz.yaml] on the Respeaker Lite for the ESP32
Using [respeaker_lite_i2s_dfu_firmware_v1.0.9.bin) on the XMOS

`[22:29:25][I][respeaker_lite:076]: Mic mute state: 0
[22:29:25][D][binary_sensor:036]: 'mute_state': Sending state OFF
[22:29:25][D][switch:055]: 'Mic mute': Sending state OFF
[22:29:25][D][media_player:080]: 'Media Player' - Setting
[22:29:25][D][media_player:093]:  Announcement: yes
[22:29:25][D][nabu_media_player.pipeline:174]: Reading FLAC file type
[22:29:25][D][nabu_media_player.pipeline:186]: Decoded audio has 1 channels, 48000 Hz sample rate, and 16 bits per sample
[22:29:25][D][nabu_media_player.pipeline:208]: Converting the audio sample rate
[22:29:25][D][nabu_media_player.pipeline:211]: Converting mono channel audio to stereo channel audio
[22:29:25][D][ring_buffer:034][speaker_task]: Created ring buffer with size 16384
[22:29:25][D][esp-idf:000][speaker_task]: I (1436637) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=4

[22:29:25][D][i2s_audio.speaker:118]: Starting Speaker
[22:29:25][D][i2s_audio.speaker:123]: Started Speaker
[22:29:28][D][esp-idf:000][speaker_task]: I (1439158) I2S: DMA queue destroyed

[22:29:28][D][i2s_audio.speaker:130]: Stopping Speaker
[22:29:28][D][i2s_audio.speaker:136]: Stopped Speaker
[22:29:29][D][switch:012]: 'Mic mute' Turning ON.
[22:29:29][D][switch:012]: 'mute_toggle' Turning ON.
[22:29:29][D][switch:055]: 'mute_toggle': Sending state ON
[22:29:30][D][switch:016]: 'mute_toggle' Turning OFF.
[22:29:30][D][switch:055]: 'mute_toggle': Sending state OFF`

The hardware switch does reflect the change in HA.

formatBCE, 2024-12-14:
Did you solder mute button to D3 pin?

deadphyre, 2024-12-14:
Well, it's possible there isn't a good enough connection, I'll try a
resolder tomorrow and update here. Thanks for the quick reply

formatBCE, 2024-12-14:
No problem. :)
The symptoms are like it's not soldered. I had it on test boards, hense the question.  :)

deadphyre, 2024-12-14:
The previous soldering job was insufficient, after resoldering the connection from Mic to the D3 pin, was able to get the Mic Mute switch in Home Assistant to work as intended.

closed 2024-12-14 by deadphyre