formatBCE/Respeaker-Lite-ESPHome-integration #28: No Text-To-Speech reply

Axydas, 2025-01-11:
Hello and thanks for the creating of the Respeaker Projects. 

I have a problem with both respeakers i have. They can listen, they are fast and they proceed with the action but the text to speech is not working at all. 

I used the 48kHz yaml code from your Github in order to flash them (of course before I flashed the latest 48kHz firmware). From Music Assistant i can play music normally.

Just note that I have also other voice assistant esp32 speakers that work perfectly with piper as text to speech (but are very much slower and not accurate). So piper is working fine. The problem is with the code of the yaml.

Any suggestion what i am doing wrong?

formatBCE, 2025-01-11:
It all looks like you didn't add resampler for the microphone.
Please double check the external_components section.

Axydas, 2025-01-11:
microphone is ok and works fine

i cannot hear the response from the speaker.

formatBCE, 2025-01-11:
> microphone is ok and works fine
> 
> i cannot hear the response from the speaker. 

Oh damn, sorry, I guess I'm too distracted today....

Do you by chance have https Home Assistant?

Axydas, 2025-01-11:
yes

formatBCE, 2025-01-11:
> yes

There's ongoing issue conversation on HA Discord about TTS proxy not working well with https and ESPHome. So far there's no solution, except setting http internal HA U$L in its settings, and use https via Nginx instead of direct certificate.

Axydas, 2025-01-11:
Yes i know. I have another speaker using many components. The tts works good with this speaker but there is noice from the cables and it is very slow

Reaspeker is much faster but the tts is not working at all.

I have to review the code.

I will try the original Seed yaml to check if I have the same result

formatBCE, 2025-01-11:
> Yes i know. I have another speaker using many components. The tts works good with this speaker but there is noice from the cables and it is very slow
>
> Reaspeker is much faster but the tts is not [...]

It won't have same issue. The problem is with Nabu media_player implementation and the TTS proxy that they've made for new generation of voice assistants.

Same trouble is with PE, and with FPH Satellite1 - basically with every ESP32-S3 satellite, that is using Nabu media_player.

Axydas, 2025-01-12:
hello again. yes you are right. the problem is with the Nabu media_player. i installed the SeeedStudio yaml and speech to text is working but the whole installation is slow and cannot understand almost anything.

With your yaml device is working quite good and much much faster but with out speech. I am not aware about this media player. do i need subscription to nabu casa? 

also i want to try your old configuration. Is it based on 48 or 16kHz firmware?

formatBCE, 2025-01-12:
Nabu media player is just a kind of media player that can be set up in ESPHome. They implemented that platform while working on PE device, and it's slowly merging into main ESPHome as we speak. 
This platform allows better integration with HA for voice assistant. It's not something that you can easily change, or add to custom project, at least so far...

formatBCE, 2025-01-12:
Also which old configuration you mean? From old_configs folder? That one is written for some INMP microphone and is just a leftover. I doubt it'll work... :(

Axydas, 2025-01-13:
> > yes
> 
> There's ongoing issue conversation on HA Discord about TTS proxy not working well with https and ESPHome. So far there's no solution, except setting http internal HA U$L in its settings, and use https via Nginx instead of direct certificate.

This was the solution to the problem in case other people are facing the same issue.

I just add the below code to the yaml and everything is working:
http_request:
   verify_ssl: false 

Thanks for the excellent support.

formatBCE, 2025-01-13:
> This was the solution to the problem in case other people are facing the same issue.
>
> I just add the below code to the yaml and everything is working:
> http_request:
>    verify_ssl: false
>
> Thanks for [...]

Hey, thank you for this! I guess I will add these lines (commented out) to the YAML, with comment to enable them if there's problem with https. 
You're awesome!

Axydas, 2025-01-13:
Ok i will try it and let you know

formatBCE, 2025-01-14:
> it is working with this line. :)
> 
> 
> > @Axydas actually, could you try something? Add this to the platformio options in YAML, and delete (comment) your http_request section. Try it please.
> > 
> > ```
> > CONFIG_MBEDTLS_SSL_PROTO_TLS1_3: "y" 
> > ```
> 
> 

Great! No compromise to the SSL. :)

closed 2025-01-14 by formatBCE