formatBCE/Respeaker-Lite-ESPHome-integration #29: Sattelite seem to be stuck in "thinking phase"

amjadsaadeh, 2025-01-17:
Hello,

I got a presoldered ESP32 Seeed Studio Voice Assistant lite board and wanted to get it working with your config. I got a few problems (I needed to adjust the wifi output_power), but finally I got it at least coinnecting to my Homeassistant instance. However, right after connection I got a sound and the LED on to is blinking blue (I assume that's fast pulse), indicating it is stuck in the thinking phase from the beginning. Removing power and restart does not change anything. I am getting the following logs:

`INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/vasat.yaml...
INFO Updating https://github.com/esphome/home-assistant-voice-pe@dev
INFO Updating https://github.com/formatBCE/home-assistant-voice-pe@48kHz_mic_support
INFO Updating https://github.com/formatBCE/Respeaker-Lite-ESPHome-integration@main
INFO Updating https://github.com/esphome/esphome.git@pull/8018/head
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING Found and merged multiple configurations for ota platform esphome port(s) [3232]
INFO Starting log output from 192.168.178.117 using esphome API
INFO Successfully connected to respeaker-satellite @ 192.168.178.117 in 0.091s
INFO Successful handshake with respeaker-satellite @ 192.168.178.117 in 0.080s
[14:18:37][I][app:100]: ESPHome version 2024.12.4 compiled on Jan 17 2025, 13:29:42
[14:18:37][I][app:102]: Project formatbce.Respeaker Lite Satellite version 2025.1.3
[14:18:37][C][wifi:600]: WiFi:
[14:18:37][C][wifi:428]:   Local MAC: F0:9E:9E:3B:3A:14
[14:18:37][C][wifi:433]:   SSID: 'FRITZ!Box 7362 SL 2GHz'[redacted]
[14:18:37][C][wifi:436]:   IP Address: 192.168.178.117
[14:18:37][C][wifi:439]:   BSSID: B0:F2:08:87:88:01[redacted]
[14:18:37][C][wifi:441]:   Hostname: 'respeaker-satellite'
[14:18:37][C][wifi:443]:   Signal strength: -54 dB ▂▄▆█
[14:18:37][V][wifi:445]:   Priority: 0.0
[14:18:37][C][wifi:447]:   Channel: 6
[14:18:37][C][wifi:448]:   Subnet: 255.255.255.0
[14:18:37][C][wifi:449]:   Gateway: 192.168.178.1
[14:18:37][C][wifi:450]:   DNS1: 192.168.178.1
[14:18:37][C][wifi:451]:   DNS2: 0.0.0.0
[14:18:37][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:37][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:37][W][respeaker_lite:069]: Read mic mute state failed
[14:18:37][C][logger:185]: Logger:
[14:18:37][C][logger:186]:   Level: VERY_VERBOSE
[14:18:37][C][logger:188]:   Log Baud Rate: 115200
[14:18:37][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[14:18:37][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:37][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:37][W][respeaker_lite:069]: Read mic mute state failed
[14:18:37][C][i2c.idf:075]: I2C Bus:
[14:18:37][C][i2c.idf:076]:   SDA Pin: GPIO5
[14:18:37][C][i2c.idf:077]:   SCL Pin: GPIO6
[14:18:37][C][i2c.idf:078]:   Frequency: 400000 Hz
[14:18:37][C][i2c.idf:084]:   Recovery: bus successfully recovered
[14:18:37][I][i2c.idf:094]: Results from i2c bus scan:
[14:18:37][I][i2c.idf:100]: Found i2c device at address 0x18
[14:18:37][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:37][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:37][W][respeaker_lite:069]: Read mic mute state failed
[14:18:37][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:37][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:37][W][respeaker_lite:069]: Read mic mute state failed
[14:18:37][VV][esp32_rmt_led_strip:101]: Writing RGB values to bus...
[14:18:37][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:37][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:37][W][respeaker_lite:069]: Read mic mute state failed
[14:18:37][C][switch.gpio:068]: GPIO Switch 'mute_toggle'
[14:18:37][C][switch.gpio:090]:   Restore Mode: always OFF
[14:18:37][C][switch.gpio:031]:   Pin: GPIO4
[14:18:37][VV][esp32_rmt_led_strip:101]: Writing RGB values to bus...
[14:18:37][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:37][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:37][W][respeaker_lite:069]: Read mic mute state failed
[14:18:37][C][esp32_rmt_led_strip:187]: ESP32 RMT LED Strip:
[14:18:37][C][esp32_rmt_led_strip:188]:   Pin: 1
[14:18:37][C][esp32_rmt_led_strip:189]:   Channel: 2
[14:18:37][C][esp32_rmt_led_strip:214]:   RGB Order: GRB
[... 642 lines cut ...]
[14:18:39][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:39][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:39][W][respeaker_lite:069]: Read mic mute state failed
[14:18:39][VV][esp32_rmt_led_strip:101]: Writing RGB values to bus...
[14:18:39][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:39][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:39][W][respeaker_lite:069]: Read mic mute state failed
[14:18:39][VV][esp32_rmt_led_strip:101]: Writing RGB values to bus...
[14:18:39][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:39][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:39][W][respeaker_lite:069]: Read mic mute state failed
[14:18:39][VV][esp32_rmt_led_strip:101]: Writing RGB values to bus...
[14:18:39][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:39][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:39][W][respeaker_lite:069]: Read mic mute state failed
[14:18:39][VV][esp32_rmt_led_strip:101]: Writing RGB values to bus...
[14:18:39][VV][i2c.idf:197]: 0x42 TX F18101
[14:18:39][VV][i2c.idf:173]: 0x42 RX 0281
[14:18:39][W][respeaker_lite:069]: Read mic mute state failed
[14:18:39][VV][esp32_rmt_led_strip:101]: Writing RGB values to bus...
`

Accoding to Homeassistant the XMOS Firmware is 1.0.8.

Do you have any clue what might be wrong?

formatBCE, 2025-01-17:
Did you solder USR and MUTE buttons to their corresponding pins?

formatBCE, 2025-01-17:
> > Did you solder USR and MUTE buttons to their corresponding pins?
> 
> I ordered a presoldered version, but I will check. Thank you.

Presoldered version has chip on board. But instructions clearly say that you have to connect onboard buttons to the pins, so they're manageable. :)
Check https://github.com/formatBCE/Respeaker-Lite-ESPHome-integration?tab=readme-ov-file#what-to-do-with-it, P2.
Without that the action (USR) button is "floating", and it is launching your assistant multiple times per second.

amjadsaadeh, 2025-01-17:
Ahh ok, I should've read the manual. Sorry.

formatBCE, 2025-01-17:
> Ahh ok, I should've read the manual. Sorry.

No problem at all! Congrats on good device! :)