formatBCE/Respeaker-Lite-ESPHome-integration #31: Wake sound/electronic noise is recorded for STT

jonathanarchergithub, 2025-02-15:
I have the issue that my STT model (big German model) often adds the word "ich" ("I") at the beginning of the recognised text which of course breaks most intents. Listening to the debug recordings I notice that the audio quality is really good, however the wake sound is heared as a loud noise at the beginning of the recording. Deactivating it doesn't solve the issue, however I notice that still some "electronic buzz" that sounds like turning on a switch can be heard at the beginning.

Is there any way to change the beginning of the audio stream capture after that initial activation noise?

formatBCE, 2025-02-15:
Hi!
How are you getting STT audio?

jonathanarchergithub, 2025-02-15:
Hi,
via configuration.yaml entry
```
assist_pipeline:
   debug_recording_dir: xxx
```

formatBCE, 2025-02-15:
> Hi,
> via configuration.yaml entry
> ```
> assist_pipeline:
>    debug_recording_dir: xxx
> ```

Let me do that to check.

formatBCE, 2025-02-15:
Okay, I can confirm the end of wake sound heard on the record. Also without wake sound there's pretty loud squeak  on start. I will reach out to Seeed devs - probably they'll help to mitigate issue.

It didn't give me any troubles with recognition though. I'm using NabuCasa for STT. It will of course depend on the model quality...

formatBCE, 2025-02-16:
Just for the info: at least FPH SAT1 has the same sound in speech record.

formatBCE, 2025-02-20:
@jonathanarchergithub i just updated my fork for `nabu_microphone`. Guys from Seeed helped me to mitigate the issue quite a bit.
Can you re-install (with caches cleaning) and try now please? For me it works much better! Thanks.

jonathanarchergithub, 2025-02-21:
Great, will try it out. Just to be certain: do I also need to update the firmware to 1.1.0 or is that related to a different issue?

jonathanarchergithub, 2025-02-23:
@formatBCE I just tested with the FW and config. Unfortunately the behaviour is still the same for me, i.e. the wake sound is fully audible in the beginning...

formatBCE, 2025-02-23:
> @formatBCE I just tested with the FW and config. Unfortunately the behaviour is still the same for me, i.e. the wake sound is fully audible in the beginning...

Well the wake sound isn't something you can avoid. It's either that, or you have to wait more before starting to listen. But the wake sound shouldn't be a problem for STT. If it is, then it's STT problem.

You may turn wake sound off and try too.

Anyways, there's nothing to do from my side, unfortunately.

closed 2025-02-23 by formatBCE

jonathanarchergithub, 2025-02-23:
Hm. Theo I don’t quite understand what you meant with the mitigation of the issue. However I agree that it shouldn’t be a big issue for STT. On the other hand it seems to miss the point of all those noise suppression/AI/whatever features that are advertised by Seeed if that wake sound is so dominant in the audio…

formatBCE, 2025-02-23:
> Hm. Theo I don’t quite understand what you meant with the mitigation of the issue. However I agree that it shouldn’t be a big issue for STT. On the other hand it seems to miss the point of all those [...]

New firmware removed "scree" sound on audio start, when no wake word sound there.

I believe the XMOS chip is not as good as they advertise, that's for sure. This is the issue for all - PE, SAT1 and Respeaker.

jonathanarchergithub, 2025-02-23:
@formatBCE Just tried without the wake sound and actually I still hear the electronic sound in the beginning as before. Wouldn't describe it as "scree" though. Sounds more like a short buzz.

formatBCE, 2025-02-23:
> @formatBCE Just tried without the wake sound and actually I still hear the electronic sound in the beginning as before. Wouldn't describe it as "scree" though. Sounds more like a short buzz.

Yup, it was much louder.

formatBCE, 2025-02-23:
Anyways, it's nothing to do with current repo. We can continue conversation with Seeed on their Discord.