formatBCE/Respeaker-Lite-ESPHome-integration #33: Timers: Intent Failed

Stepyon, 2025-02-21:
Hi Andrii,

First of all, thank you for this repo! I reused your sources for my ReSpeaker-Lite firmware, and almost everything works perfectly.

The only issue I'm having is with setting timers, and it’s probably on my side. When I set a timer, I don’t get any voice confirmation, and I see the following error (in red) in the logs:

[image]

The timer itself does get set and rings, but there’s no voice confirmation. I’m using my own ESPHome configuration based on yours—modified for my device and my needs (attached).

For voice recognition I use the Vosk add-on with the Russian language model (which actually works very well) and Gemini API as the Assistant LLM. Any help or hints would be greatly appreciated!

[respeaker-s3-cfg.zip](https://github.com/user-attachments/files/18915467/respeaker-s3-cfg.zip)

formatBCE, 2025-02-21:
Hey!
Yeah that's something strange, and i'm not sure it's the YAML. It might be the pipeline.
Here's my log from similar call:
```
[13:32:02][D][voice_assistant:515]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[13:32:02][D][voice_assistant:515]: State changed from AWAITING_RESPONSE to AWAITING_RESPONSE
[13:32:02][D][voice_assistant:641]: Event Type: 4
[13:32:02][D][voice_assistant:669]: Speech recognised as: "Set a timer for 60 seconds."
[13:32:02][D][voice_assistant:641]: Event Type: 5
[13:32:02][D][voice_assistant:674]: Intent started
[13:32:03][D][voice_assistant:843]: Timer Event
[13:32:03][D][voice_assistant:844]:   Type: 0
[13:32:03][D][voice_assistant:845]:   Timer(id=01JMN6SECGGZGKC1YSZQA55K2T, name=, total_seconds=60, seconds_left=60, is_active=YES)
[13:32:03][D][light:036]: 'voice_assistant_leds' Setting:
[13:32:03][D][light:051]:   Brightness: 66%
[13:32:03][D][voice_assistant:641]: Event Type: 6
[13:32:03][D][voice_assistant:641]: Event Type: 7
[13:32:03][D][voice_assistant:697]: Response: "Timer set for 60 seconds"
[13:32:03][D][light:036]: 'voice_assistant_leds' Setting:
[13:32:03][D][light:051]:   Brightness: 66%
[13:32:03][D][light:109]:   Effect: 'Replying'
[13:32:03][D][voice_assistant:641]: Event Type: 8
[13:32:03][D][voice_assistant:719]: Response URL: "http://192.168.1.173:8123/api/tts_proxy/sDYRgQXTXTyJau2FGX2Vwg.flac"
[13:32:03][D][voice_assistant:515]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
```
See, after timer event, i'm getting Event 6 and Event 7, but your log shows Event 0. So something, probably, is wrong with TTS engine - it tries to generate response and fails. The error "unexpected error during intent recognition" is often to be seen in voice assistant pipeline debug. I'd check text-to-speech logs.

Stepyon, 2025-02-21:
Yeah, it is pipeline, checked the logs (attached) and it looks like the issue is with how the STT service provides numbers on my side. Home Assistant’s built-in timer domain expects numeric digits (e.g., "60"), but my STT (Vosk) generates spelled-out text (e.g., "sixty" or "five").

Home Assistant tries to parse "пять" as an integer when tries to generate a reply using the template and fails with an error:

> ValueError: Template error: int got invalid input 'пять' when rendering template

Now I need to find a way how to fix this...

[HA logs.txt](https://github.com/user-attachments/files/18916528/HA.logs.txt)

formatBCE, 2025-02-21:
Hopefully it will be fixed!

closed 2025-02-21 by formatBCE