formatBCE/Respeaker-Lite-ESPHome-integration #34: Regarding the error issue of respeaker-satellite-base.yaml

cuichaobaobao, 2025-02-22:
I followed the installation steps and flashed the firmware. My ReSpeaker Lite is working properly, but when I view the code interface from the HA panel at /config/esphome/common/respeaker-satellite-base.yaml, it shows an error. I don’t quite understand the reason behind this. Thank you very much!

[image]

[image]

formatBCE, 2025-02-22:
File Editor sometimes can give weird false errors. The best indicator of working YAML is working device. You can also check in ESPHome dashboard - should be no errors there.

closed 2025-02-22 by formatBCE

cuichaobaobao, 2025-02-22:
Got it, thanks! My ReSpeaker Lite is running fine now. So I can ignore this error message?

formatBCE, 2025-02-22:
> Got it, thanks! My ReSpeaker Lite is running fine now. So I can ignore this error message?

Yup you surely can.

cuichaobaobao, 2025-02-23:
Okay, thank you very much! Also, I noticed that you updated it again. Do I need to flash the firmware again, or should I wait for a more stable version? Since I have to reflash it every time there’s an update. 😄

formatBCE, 2025-02-23:
> Okay, thank you very much! Also, I noticed that you updated it again. Do I need to flash the firmware again, or should I wait for a more stable version? Since I have to reflash it every time there’s an update. 😄

You can skip it for now, if you have 2025.2.1. The new version 2025.2.2 brings auto-updates to Respeaker board (you won't need to disassemble the device and update through dfu-util. But if you already updated XMOS to 1.1.0, you can skip.

cuichaobaobao, 2025-02-23:
I checked, and mine is still version 2.0, with XMOS still at version 1.0.9. I’ll go ahead and flash the latest firmware again.

[image]

formatBCE, 2025-02-23:
> I checked, and mine is still version 2.0, with XMOS still at version 1.0.9. I’ll go ahead and flash the latest firmware again.
> 
> [image]

Nice! You will see update process then :)

cuichaobaobao, 2025-02-23:
Thanks! I’ll try reflashing the firmware later.

formatBCE, 2025-02-23:
> Thanks! I’ll try reflashing the firmware later.

You can do it wirelessly from ESPHome, you know it right?

cuichaobaobao, 2025-02-23:
Yep, I’m aware of that.🙏