formatBCE/Respeaker-Lite-ESPHome-integration #38: Microphone MWW Sensitivity

chevellebro1, 2025-02-26:
I recently updated my Respeaker devices. I have the newest DFU and updated my ESP yaml. I notice that now I have to yell in order to get the device to trigger whereas before I could speak at a normal volume. I've tried adjusting the amplify_shift to 8 (max) and it makes no difference. If I switch the MWW microphone back to channel 0 I get the sensitivity back but it does not pick up my voice command...I'm probably missing something but hopefully you can point me in the right direction. I've tried on two devices and even tried updating the DFU from the Respeaker_Lite repository but no difference.

formatBCE, 2025-02-26:
Huh. For me it's better than before.
Can you try changing the mww model to default "ok_nabu", and put probability_cutoff lower? Probably it's not the mics, but the new model doesn't fit you?

chevellebro1, 2025-02-26:
It’s similar where I have to get very close to the ReSpeaker device for it to respond even using Okay Nabu model. I used to be able to be across the room without an issue. When I set the microphone for mww to channel 0 it does pick me up as if did before

formatBCE, 2025-02-26:
Huh... I guess you'll need to revert to 1.0.9 and old channels.

chevellebro1, 2025-02-26:
Thanks I reverted and no issue.

formatBCE, 2025-02-26:
I will watch it closely. And let Seed guys know, that at least for some users it gets worse with 1.1.0.

Stepyon, 2025-02-26:
I feel like I also have this issue, especially "stop" command is now recognized after about 5-8 attempts which is not usable. I didn't try to revert 1.0.9 so can't confirm for sure it is 1.1.0 related.

formatBCE, 2025-03-13:
Guys, try new config. I realized that VAD was cutting off a lot of wake word attempts. I lowered cut-off probability to 5% instead of default 50%.

Stepyon, 2025-03-13:
Yes, I noticed in the logs that VAD was cutting off wake words, but I didn't fully understand exactly what the VAD model is, what it does, or how it relates to wake-word models. Setting the cutoff to 0.05 seems to help, and it's working much better now. However, could this change also increase false triggers?

mikey60, 2025-03-26:
My MWW response has been poor and varied.   I seem to have much a improved response when I changed line 23 in the nabu_microphone.cpp (48K version) 
**from:**    static const size_t DMA_BUFFER_SIZE = 512;
**to:**        static const size_t DMA_BUFFER_SIZE = 480;
I have switched back and forth a couple of times and 480 seems to work much better for me.  480 is exactly 3X of the 160 value that VPE uses to account for the 48KHz to 16KHz ratio.  I originally rounded up slightly to 512 thinking is was better because 512 can be factored by 32.  Maybe there is some interaction of the DMA buffer size MMW or maybe it is something weird with my device/configuration.

formatBCE, 2025-03-26:
> My MWW response has been poor and varied.   I seem to have much a improved response when I changed line 23 in the nabu_microphone.cpp (48K version)
> **from:**    static const size_t DMA_BUFFER_SIZE [...]

Thanks for reporting! I will test with 480, and update mainstream.

mikey60, 2025-03-26:
I forgot to mention that I am using the "Hey Jarvis" wake word.  Though "stop" also worked better.

formatBCE, 2025-03-27:
Tested it. I definitely see improvement in Ok Nabu, and in "stop" words. Thank you @mikey60 , you're great (again) :)

mikey60, 2025-03-27:
I apologize for not testing more thoroughly the first time.

formatBCE, 2025-03-27:
Huh, it was still good, right? Improvements are always great!

Stepyon, 2025-03-28:
> Tested it. I definitely see improvement in Ok Nabu, and in "stop" words. Thank you [@mikey60](https://github.com/mikey60) , you're great (again) :)

Is this limited to wake words only, or does it also enhance voice recording and speech recognition through STT?

formatBCE, 2025-03-28:
> > Tested it. I definitely see improvement in Ok Nabu, and in "stop" words. Thank you [@mikey60](https://github.com/mikey60) , you're great (again) :)
> 
> Is this limited to wake words only, or does it also enhance voice recording and speech recognition through STT?

I'm not 100% sure, but it should improve everything related to the microphones.

closed 2025-04-10 by formatBCE