formatBCE/Respeaker-Lite-ESPHome-integration #39: HA fails to announce to the satellite

TheMangrove, 2025-03-07:
The speaker seems to work and accepts voice commands and will play sounds, but none of the speech to text plays over the speaker. 

Additionally, if I try and "Set up Voice Assistant" in HA, I get the following error from HA "The voice assistant is unable to connect to Home Assistant". From the ESPHome logs though it appears to connect just fine, but rather HA doesn't seem to connect to the speaker. 

When I try and "Set up voice assistant" in HA, I get the following Errors in the ESPHome logs:
```
[23:40:08][E][speaker_media_player:334]: The announcement pipeline's file reader encountered an error.
[23:40:12][D][esp-idf:000][ann_read]: E (524247) esp-tls: [sock=58] select() timeout
[23:40:12][D][esp-idf:000][ann_read]: E (524249) transport_base: Failed to open a new connection: 32774
[23:40:12][D][esp-idf:000][ann_read]: E (524251) HTTP_CLIENT: Connection failed, sock < 0
[23:40:13][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_ERR_HTTP_CONNECT
```

I have tried uncommenting 
```
 http_request:
   verify_ssl: false
```

I have tried the troubleshooting steps that the HA error links to with no success. 

I am running HA/ESPHome and all other packages as a Docker container using the latest builds. Ingress is handled through a custom domain using Traefik. 

Any help would be appreciated.

formatBCE, 2025-03-07:
Hi!
First, restart your HA and TTS/STT (if using local).
Second, make sure HA and satellite are in same subnet.
Third, after running command, go to voice pipeline debug, and try playing response audio there.

If the problem is elsewhere, we'll need your logs from satellite...

TheMangrove, 2025-03-07:
Thanks for the response.

>First, restart your HA and TTS/STT (if using local).

Done. No change. 

>Second, make sure HA and satellite are in same subnet.

They are but that's a bit more complicated. I'll preface I'm not an expert in networking, but I'll do my best to explain the set up as I understand it. I'm still learning Docker as well as ingress.

So all the HA Docker containers are running through Truenas Scale. Both the server (which hosts all the docker containers) and the voice satellite are on the same subnet with static IP's. However, ingress to the Docker containers are configured through Traefik to use my custom domain "container.mydomain.com" which is supplied/redirected by custom DNS settings in my Pihole.  Traefik then redirects to the appropriate internal Docker network. My routers DHCP and assigned static IP's are all 192.168.1.X while the internal Docker network seems to use 172.16.3.X. This, I suspect is where the issue lies, however I'm not sure why TTS won't play to the speaker as HA is clearly connected to the speaker. The HA containers, voice pipeline etc are all able to access each other internally.

I also have a wyoming satellite using a Pi 2 Zero with the respeaker 2 mic hat working as a voice assistant without issue and STT/TTS work no problem. It's connected through the Wyoming protocol though and not through ESPHome. I have other ESPHome devices and all work just fine so there is something unique about this specific device that is causing an issue. 

>Third, after running command, go to voice pipeline debug, and try playing response audio there.

Done. Audio response plays through my computer speaker just fine and as far as the voice assistant pipline is concerned it all seems to work, it just didn't manage to play the response through the speaker. Additionally, if I go in through the built in media player and try and play something, HA thinks it played but nothing comes through the speaker and ESPHome logs don't register anything. 

So, as far as I can tell, internally everything works, and the Voice Assistant communicates just fine _to_ the server, but for some reason the server is unable to, or fails to send audio back to the Voice Assistant. 

Please let me know if there is anything else I can provide to help troubleshoot this.

formatBCE, 2025-03-07:
Hmm, well. If the sounds of activation, mute/unmute and timer are playing too - that means only access to the FLAC over network is broken.

Even if ESPHome can connect, the sound from HA to ESPHome satellites (not to Wyoming, it's ESPHome specific!) goes through so-called sound proxy. That proxy is exposing files through HA external address. 

Do two things please.
1. Go to Settings - System - Network and check "Home Assistant URL" section there, specifically Local value.
2. In ESPHome logs, you should see full URL to the FLAC file when it tries to play it. Check that URL, and try also copy it and paste into the browser to see if it's accessible.

Thanks.

TheMangrove, 2025-03-07:
>Go to Settings - System - Network and check "Home Assistant URL" section there, specifically Local value.

Currently the url is set to the 172.16.3.x:8123 which is the internal Docker IP. 

I have tried setting it to "automatic" as well as "https://homeassistant.mydomain.com" which is how I connect to HA through a browser. Neither have worked. It should be noted that the 172.16.3.x:8123 IP is not accessible externally (if I type that into my browser I get a 404). This is the IP that traefik redirects to when someone tries to access "https://homeassistant.mydomain.com"

>In ESPHome logs, you should see full URL to the FLAC file when it tries to play it. Check that URL, and try also copy it and paste into the browser to see if it's accessible.

`[00:44:07][D][text_sensor:064]: 'TTS URI': Sending state 'http://172.16.3.7:8123/api/esphome/ffmpeg_proxy/77feebfa3d29b26de3ab168456db744f/abGFzJYW-VuxBMX2Ip1_jw.flac'`

So this is not accessible if I try and past it in the browser as the IP is the internal Docker one. 

So am I correct to assume that HA doesn't send or stream the flac file directly, but rather tells the voice assistant to access and play the file at "X" location? This would then be the crux of the issue since it's telling it to access an internal IP which wouldn't be accessible to the device. 

In that case, how does the wyoming protocol stream audio to the voice assistant that's using the RPI? Is this doing it in a fundamentally different way, or is there something in that setup that might point to a simple solution for the ESPHome device? 

The simplest solution would be to somehow expose the internal docker IP externally but I'll have to look into how to do that properly without messing up all the traefik settings.

TheMangrove, 2025-03-07:
From what I can tell/remember when I set it up, Wyoming uses MQTT which would make sense and explain why it doesn't have this same issue. So I don't think looking to Wyoming for a solution will work since it's fundamentally different.

formatBCE, 2025-03-07:
You have your reason.
Wyoming satellite gets direct link to TTS audio file. But for ESPHome satellites HA does conversion to FLAC and plays proxy role.
You may put local IP there in HA settings, with port 8123. It should be accessible via plain HTTP. It will resolve problem.

TheMangrove, 2025-03-07:
OK, Thanks. That helped me narrow on the issue. It was actually quite simple in that port 8123 wasn't open on the server for some reason.  Opening that and then changing the local network to my static IP solved the issue. Thanks for all your help.

closed 2025-03-07 by TheMangrove