formatBCE/Respeaker-Lite-ESPHome-integration #42: After pasting yaml geting error :

automation-agn, 2025-03-27:
`INFO ESPHome 2025.3.2
INFO Reading configuration /config/esphome/koala-satellite.yaml...
INFO Updating https://github.com/esphome/home-assistant-voice-pe@dev
INFO Updating https://github.com/formatBCE/home-assistant-voice-pe@48kHz_mic_support
INFO Updating https://github.com/formatBCE/Respeaker-Lite-ESPHome-integration@main
Failed config

esphome: [source /config/esphome/koala-satellite.yaml:21]
  
  'name' is a required option for [esphome].
  project: 
    name: formatbce.Respeaker Lite Satellite
    version: 2025.3.1
  min_version: 2025.3.2
  on_boot: 
    priority: 375
    then: 
      - sensor.template.publish: 
          id: next_timer
          state: -1
      - script.execute: control_leds
      - delay: 10min`
And code is highlited:

[image]

formatBCE, 2025-03-27:
You have to merge both files (base and other) if you're using one file.
Or place this one into common dir in your esphome and use other one as device config.

closed 2025-03-27 by formatBCE

elektrinis, 2025-06-07:
I'm getting same error. What do you mean, merge? What with what? Please spoonfeed us a little. Thanks.

formatBCE, 2025-06-07:
> I'm getting same error. What do you mean, merge? What with what? Please spoonfeed us a little. Thanks.

This means add the content of both files to single file, and remove lines corresponding to "common" folder.

elektrinis, 2025-06-07:
OK, I did not think that we need to merge the contents of both files. This clears things up, thanks.

formatBCE, 2025-06-07:
> OK, I did not think that we need to merge the contents of both files. This clears things up, thanks.

You may place "common"  folder into ESPHome root folder, and don't merge :)