formatBCE/Respeaker-Lite-ESPHome-integration #45: boot loop

nikscha, 2025-04-02:
I'm experiencing the following issue:

The respeaker board is starting, and for a brief moment it's available in Home Assistant. Then there's a "chirping" sound, and the respeaker restarts. 
These are the logs:

> [00:20:43]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> [00:20:43]
> [00:20:43]Core  0 register dump:
> [00:20:43]PC      : 0x42004319  PS      : 0x00060630  A0      : 0x8202140e  A1      : 0x3fcb0ab0  
> [00:20:43]A2      : 0x3fcb0ae8  A3      : 0x00000000  A4      : 0x3fcb0af0  A5      : 0x00000000  
> [00:20:43]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x82024fab  A9      : 0x3fcb0c00  
> [00:20:43]A10     : 0x0000000b  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
> [00:20:43]A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
> [00:20:43]EXCVADDR: 0x00000000  LBEG    : 0x400554b9  LEND    : 0x400554dd  LCOUNT  : 0x82024fd9  
> [00:20:43]
> [00:20:43]
> [00:20:43]Backtrace: 0x42004316:0x3fcb0ab0 0x4202140b:0x3fcb0ae0 0x4202148d:0x3fcb0b40 0x4202503b:0x3fcb0b60 0x42025091:0x3fcb0ba0 0x42020595:0x3fcb0bc0 0x42029abd:0x3fcb0be0 0x42024f91:0x3fcb0c00 0x42024fa8:0x3fcb0c20 0x420225df:0x3fcb0c40 0x42022612:0x3fcb0c90 0x42108609:0x3fcb0cb0 0x421086d3:0x3fcb0cd0 0x42028410:0x3fcb0cf0 0x4202c476:0x3fcb0d20 0x4200e4ae:0x3fcb0d40
> [00:20:43]
> [00:20:43]
> [00:20:43]
> [00:20:43]
> [00:20:43]ELF file SHA256: 529a6935cb55cf3c
> [00:20:43]
> [00:20:43]Rebooting...
> [00:20:43]ESP-ROM:esp32s3-20210327
> [00:20:43]Build:Mar 27 2021
> [00:20:43]rst:0x3 (RTC_SW_SYS_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
> [00:20:43]Saved PC:0x4037b27b
> [00:20:43]SPIWP:0xee
> [00:20:43]mode:DIO, clock div:1
> [00:20:43]load:0x3fce3818,len:0x1750
> [00:20:43]load:0x403c9700,len:0x4
> [00:20:43]load:0x403c9704,len:0xbe4
> [00:20:43]load:0x403cc700,len:0x2d34
> [00:20:44]entry 0x403c9908
> [00:20:44]I (26) boot: ESP-IDF 5.1.5 2nd stage bootloader
> [00:20:44]I (26) boot: compile time Apr  2 2025 21:16:03
> [00:20:44]I (26) boot: Multicore bootloader
> [00:20:44]I (29) boot: chip revision: v0.2
> [00:20:44]I (33) boot.esp32s3: Boot SPI Speed : 80MHz
> [00:20:44]I (38) boot.esp32s3: SPI Mode       : DIO
> [00:20:44]I (43) boot.esp32s3: SPI Flash Size : 8MB
> [00:20:44]I (47) boot: Enabling RNG early entropy source...
> [00:20:44]I (53) boot: Partition Table:
> [00:20:44]I (56) boot: ## Label            Usage          Type ST Offset   Length
> [00:20:44]I (64) boot:  0 otadata          OTA data         01 00 00009000 00002000
> [00:20:44]I (71) boot:  1 phy_init         RF data          01 01 0000b000 00001000
> [00:20:44]I (78) boot:  2 app0             OTA app          00 10 00010000 003c0000
> [00:20:44]I (86) boot:  3 app1             OTA app          00 11 003d0000 003c0000
> [00:20:44]I (93) boot:  4 nvs              WiFi data        01 02 00790000 0006d000
> [00:20:44]I (101) boot: End of partition table
> [00:20:44]I (105) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=156a7ch (1403516) map
> [00:20:44]I (365) esp_image: segment 1: paddr=00166aa4 vaddr=3fc9db00 size=04834h ( 18484) load
> [00:20:44]I (370) esp_image: segment 2: paddr=0016b2e0 vaddr=40378000 size=04d38h ( 19768) load
> [00:20:44]I (376) esp_image: segment 3: paddr=00170020 vaddr=42000020 size=1124e8h (1123560) map
> [00:20:44]I (581) esp_image: segment 4: paddr=00282510 vaddr=4037cd38 size=10ce4h ( 68836) load
> [00:20:44]I (606) boot: Loaded app from partition at offset 0x10000
> [00:20:44]I (606) boot: Disabling RNG early entropy source...
> [00:20:44]I (607) cpu_start: Multicore app
> [00:20:44]I (611) octal_psram: vendor id    : 0x0d (AP)
> [00:20:44]I (615) octal_psram: dev id       : 0x02 (generation 3)
> [00:20:44]I (621) octal_psram: density      : 0x03 (64 Mbit)
> [00:20:44]I (627) octal_psram: good-die     : 0x01 (Pass)
> [00:20:44]I (632) octal_psram: Latency      : 0x01 (Fixed)
> [00:20:44]I (637) octal_psram: VCC          : 0x01 (3V)
> [00:20:44]I (642) octal_psram: SRF          : 0x01 (Fast Refresh)
> [00:20:44]I (648) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
> [00:20:44]I (654) octal_psram: BurstLen     : 0x01 (32 Byte)
> [00:20:44]I (659) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
> [00:20:44]I (666) octal_psram: DriveStrength: 0x00 (1/1)
[... 837 lines cut ...]
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][D][wifi:545]: - [redacted] [redacted]▂▄▆█
> [00:21:33][I][wifi:313]: WiFi Connecting to 'PickleRick2.4'...
> [00:21:33][D][esp-idf:000][wifi]: W (7404) wifi:
> [00:21:33][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
> [00:21:33][D][esp-idf:000][wifi]: 
> [00:21:33]
> [00:21:33][D][esp-idf:000][wifi]: I (7412) wifi:
> [00:21:33][D][esp-idf:000][wifi]: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
> [00:21:33][D][esp-idf:000][wifi]: 
> [00:21:33]
> [00:21:33][D][esp-idf:000][wifi]: I (7414) wifi:
> [00:21:34][D][esp-idf:000][wifi]: state: init -> auth (b0)

formatBCE, 2025-04-02:
Well looks like it's connecting to HA and failing. You will have to decode stack trace, with current log it's impossible to say what is happening. I didn't experience this, nor others before.

nikscha, 2025-04-02:
I found the issue:

I had the 'captive portal' enabled, after taking it out I'm not experiencing the issue anymore

closed 2025-04-02 by nikscha

formatBCE, 2025-04-02:
Well as the rule of thumb, if you have altered config, try resetting to default before creating issue. :)

nikscha, 2025-04-02:
> Well as the rule of thumb, if you have altered config, try resetting to default before creating issue. :)

Yeah :) 😅