formatBCE/Respeaker-Lite-ESPHome-integration #46: New Config file updated 3 days ago.

DocB224, 2025-04-04:
Can you help me out. You are the first person to provide an update (3 days ago) on the Respeaker yaml. I tried loading it and I get errors in ESP. I am new to this and having been trying for hours to get it to work. Can you provide me steps on what I need to edit in this yaml  to work for me. Also first error is device name has to many characters. I tried shortening it and that didn't work. Thanks in advance. I am using ESP Home builder.

formatBCE, 2025-04-04:
Please provide log. I'm no wizard of any kind, so willed your help. :)

Did you alter the config?

formatBCE, 2025-04-04:
```
Could not open /dev/ttyUSB0
```

This means your ESP32 isn't connected to your PC/Mac properly. Make sure you connected ESP32, not the Respeaker USB. Also you will need to hold "boot" button (tiny one to the left of USB port) while you're plugging in the USB cable.

DocB224, 2025-04-05:
Thank you for answering my email and trying to help.

Okay I started from the beginning and can't get past this. It's the
opening paragraph in the yamal file.

Error I am receiving in ESP Home device builder running the validate
command for a yaml review.

INFO ESPHome 2025.3.3 INFO Reading configuration
/config/esphome/home-assistant-voice.yaml...
INFO Updating ***@***.***
INFO Unable to import component Substitutions: No module named
'esphome.components.Substitutions' Failed config Substitutions: [source
/config/esphome/home-assistant-voice.yaml:4] Component not found:
Substitutions. voice_assist_idle_phase_id: 1
voice_assist_waiting_for_command_phase_id:
2 voice_assist_listening_for_command_phase_id: 3
voice_assist_thinking_phase_id:
4 voice_assist_replying_phase_id: 5 voice_assist_not_ready_phase_id:
10 voice_assist_error_phase_id:
11 hidden_ssid: false.

Yaml Script

Substitutions:
  # Phases of the Voice Assistant
  # The voice assistant is ready to be triggered by a wake word
  voice_assist_idle_phase_id: "1"
  # The voice assistant is waiting for a voice command (after being
triggered by the wake word)
  voice_assist_waiting_for_command_phase_id: "2"
  # The voice assistant is listening for a voice command
  voice_assist_listening_for_command_phase_id: "3"
  # The voice assistant is currently processing the command
  voice_assist_thinking_phase_id: "4"
  # The voice assistant is replying to the command
  voice_assist_replying_phase_id: "5"
  # The voice assistant is not ready
  voice_assist_not_ready_phase_id: "10"
  # The voice assistant encountered an error
  voice_assist_error_phase_id: "11"
  # Change this to true in case you ahve a hidden SSID at home.
  hidden_ssid: "false"

formatBCE, 2025-04-05:
Sorry to be rude, but all this stuff doesn't correspond to actual errors in YAML. It's something wrong with basic setup. So please check the tutorials on ESPHome, probably reinstall, ask for help on ESPHome Discord. Regarding the YAML - just make sure you have copied everything from both example and base files into your config, if you prefer one file (without "common" directory).

The issues you're getting aren't related to the repo. Closing this.

closed 2025-04-05 by formatBCE