formatBCE/Respeaker-Lite-ESPHome-integration #48: Voice Dynamic Volume

Nortonko, 2025-04-14:
Hi, can you add support for this project: [https://github.com/jaapp/ha-voice-dynamic-volume](url)? It is not compatible with your version. In koala-base.yaml in your code a different id name is used for the mic on channel 0 -> `nabu_mic_va` vs `asr_mic`.
Thanks.

formatBCE, 2025-04-14:
Hi!
So far you can use full code and adjust it - either rename mic in Koala YAML or in the additional YAML.
I will rename mic in next release, making it compatible.

Nortonko, 2025-04-14:
Hi! Yes i already did it ;-).

formatBCE, 2025-05-17:
I've prepared the changes for this - but looks like with new changes to PE, microphones configuration will be changed. I will make sure my config is identical to PE.
Closing this.

closed 2025-05-17 by formatBCE