formatBCE/Respeaker-Lite-ESPHome-integration #50: External speaker

vincino77, 2025-04-22:
Would it be possible to modify the code to give voice response on an external speaker available in Home Assistant? (eg a google home)

formatBCE, 2025-04-22:
You can do it already - there's event sent every time when satellite speaks. Just make an automation to take the URL with voice response from the event and send it to any speaker. Then mute volume on Respeaker itself.

If you need example, I can write.

vincino77, 2025-04-22:
What you suggested works, but it introduces a small delay in the execution of the response. When the continuous mode is active, the respeaker opens the microphone while the TTS is still running and hears its own voice in an infinite loop! I think I need to insert a delay somewhere, but I'm not sure where. Any ideas?

formatBCE, 2025-04-22:
Well this is another problem I guess? you mean, when playing TTS on another speaker, the delay between event and actual audio start on external speaker is enough to trigger Respeaker with the end of phrase, because it finished "speaking" earlier than external speaker?

I don't really know what to do there in this case. This is grey area already - with redirecting the response - and I don't want to alter the functionality even more... Probably some delay in voice assistant would help - but I don't know where to put it. :)

vincino77, 2025-04-22:
You totally understand the problem. If you think of anything I can do, let me know. Thanks in advance

formatBCE, 2025-04-22:
Long shot, but you could use some global boolean bar, set it to "true" in "on_tts_start" (there you have text, so you can see if it ends with "?") - and then in on_tts_end, after sending event with URI to speak on other speaker - introduce delay if that variable is true (and of course reset it to false). That should do.

vincino77, 2025-04-22:
I don't really understand what I have to do, but I'll try... thanks! :)

vincino77, 2025-04-22:
No problem. Thanks for the support, it gave me a lot of ideas. Now it's just a matter of studying a little. Thanks again

formatBCE, 2025-04-22:
That's the way! Good luck!

closed 2025-04-23 by formatBCE