formatBCE/Respeaker-Lite-ESPHome-integration #53: Announcement not playing, media not accessible

elektrinis, 2025-06-07:
home assistant integration through music assistant.
When I run tts.speak - it works well. Radio streaming works well too.
But when using assist, it recognizes the voice, carries out action, but nothing plays back. Blue LED is blinking for like 10 seconds, but no audio.

I have tried to uncomment this, but it did not help:
```
http_request:
  verify_ssl: false
```

I have followed the URL to flac media, and it does not load.
I have also tried to change `announcement_pipeline:` and `media_pipeline:` to MP3, but same result.
What may be the issue?
I'm using microsoft azure for tts (since it's the only one that supports my language).

log:

```
[16:34:39][D][micro_wake_word:325]: Detected 'Okay Nabu' with sliding average probability is 0.87 and max probability is 0.98
[16:34:39][D][media_player:074]: 'voice-speaker' - Setting
[16:34:39][D][media_player:078]:   Command: STOP
[16:34:39][D][media_player:087]:  Announcement: yes
[16:34:39][D][speaker_media_player:408]: State changed to ANNOUNCING
[16:34:39][D][ring_buffer:034][ann_read]: Created ring buffer with size 1000000
[16:34:39][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[16:34:39][D][speaker_media_player.pipeline:124]: Decoded audio has 1 channels, 48000 Hz sample rate, and 16 bits per sample
[16:34:39][D][ring_buffer:034]: Created ring buffer with size 9600
[16:34:39][D][voice_assistant:456]: State changed from IDLE to START_MICROPHONE
[16:34:39][D][voice_assistant:463]: Desired state set to START_PIPELINE
[16:34:39][D][voice_assistant:186]: Starting Microphone
[16:34:39][D][ring_buffer:034]: Created ring buffer with size 16384
[16:34:39][D][voice_assistant:456]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[16:34:39][D][voice_assistant:456]: State changed from STARTING_MICROPHONE to START_PIPELINE
[16:34:39][D][voice_assistant:207]: Requesting start...
[16:34:39][D][voice_assistant:456]: State changed from START_PIPELINE to STARTING_PIPELINE
[16:34:39][D][voice_assistant:478]: Client started, streaming microphone
[16:34:39][D][voice_assistant:456]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[16:34:39][D][voice_assistant:463]: Desired state set to STREAMING_MICROPHONE
[16:34:39][D][voice_assistant:598]: Event Type: 1
[16:34:39][D][voice_assistant:601]: Assist Pipeline running
[16:34:39][D][voice_assistant:598]: Event Type: 3
[16:34:39][D][voice_assistant:612]: STT started
[16:34:39][D][light:036]: 'led_internal' Setting:
[16:34:39][D][light:051]:   Brightness: 100%
[16:34:39][D][light:058]:   Red: 100%, Green: 20%, Blue: 100%
[16:34:39][D][light:109]:   Effect: 'Slow Pulse'
[16:34:40][D][speaker_media_player:408]: State changed to PLAYING
[16:34:40][D][voice_assistant:598]: Event Type: 11
[16:34:40][D][voice_assistant:758]: Starting STT by VAD
[16:34:40][D][light:036]: 'led_internal' Setting:
[16:34:40][D][light:051]:   Brightness: 100%
[16:34:40][D][light:058]:   Red: 100%, Green: 20%, Blue: 100%
[16:34:42][D][voice_assistant:598]: Event Type: 12
[16:34:42][D][voice_assistant:762]: STT by VAD end
[16:34:42][D][voice_assistant:456]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[16:34:42][D][voice_assistant:463]: Desired state set to AWAITING_RESPONSE
[16:34:42][D][voice_assistant:456]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[16:34:42][D][light:036]: 'led_internal' Setting:
[16:34:42][D][light:051]:   Brightness: 100%
[16:34:42][D][light:058]:   Red: 100%, Green: 20%, Blue: 100%
[16:34:42][D][light:109]:   Effect: 'Fast Pulse'
[16:34:42][D][voice_assistant:456]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[16:34:42][D][voice_assistant:598]: Event Type: 4
[16:34:42][D][voice_assistant:626]: Speech recognised as: "Išjunk šviesą miegamajame."
[16:34:42][D][voice_assistant:598]: Event Type: 5
[16:34:42][D][voice_assistant:631]: Intent started
[16:34:45][D][voice_assistant:598]: Event Type: 6
[16:34:45][D][voice_assistant:598]: Event Type: 7
[16:34:45][D][voice_assistant:656]: Response: "Atlikta."
[16:34:45][D][light:036]: 'led_internal' Setting:
[16:34:45][D][light:051]:   Brightness: 100%
[16:34:45][D][light:058]:   Red: 20%, Green: 100%, Blue: 100%
[16:34:45][D][light:109]:   Effect: 'Slow Pulse'
[16:34:45][D][voice_assistant:598]: Event Type: 8
[16:34:45][D][voice_assistant:678]: Response URL: "https://my.domain.com/api/tts_proxy/bgLTnd8TKbdwB6bptg7ENQ.flac"
[16:34:45][D][voice_assistant:456]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[16:34:45][D][voice_assistant:463]: Desired state set to STREAMING_RESPONSE
[16:34:45][D][media_player:074]: 'voice-speaker' - Setting
[16:34:45][D][media_player:081]:   Media URL: https://my.domain.com/api/tts_proxy/bgLTnd8TKbdwB6bptg7ENQ.flac
[16:34:45][D][media_player:087]:  Announcement: yes
[16:34:45][D][speaker_media_player:408]: State changed to ANNOUNCING
[16:34:45][D][voice_assistant:598]: Event Type: 2
[16:34:45][D][voice_assistant:697]: Assist Pipeline ended
[16:34:51][D][esp-idf:000][ann_read]: E (145850) transport_base: esp_tls_conn_read error, errno=No more processes
[16:34:51][D][esp-idf:000][ann_read]: W (145850) HTTP_CLIENT: Connection timed out before data was ready!
[16:34:51][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[16:34:51][E][speaker_media_player:328]: The announcement pipeline's file reader encountered an error.
[16:34:57][D][esp-idf:000][ann_read]: E (151919) transport_base: esp_tls_conn_read error, errno=No more processes
[16:34:57][D][esp-idf:000][ann_read]: W (151919) HTTP_CLIENT: Connection timed out before data was ready!
[16:34:57][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[16:34:57][E][speaker_media_player:328]: The announcement pipeline's file reader encountered an error.
[16:34:57][D][speaker_media_player:408]: State changed to PLAYING
[16:34:57][D][voice_assistant:329]: Announcement finished playing
[16:34:57][D][voice_assistant:456]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[16:34:57][D][voice_assistant:463]: Desired state set to RESPONSE_FINISHED
[16:34:57][D][voice_assistant:456]: State changed from RESPONSE_FINISHED to IDLE
[16:34:57][D][voice_assistant:463]: Desired state set to IDLE
```

elektrinis, 2025-06-07:
I've switched to internal url, now it looks like this:

`http://192.168.69.4/api/tts_proxy/e2ELzasSf5q4JHfuG4HZkw.mp3`

still not loading. I understand it's offtopic, but do you have any ideas?

formatBCE, 2025-06-07:
It looks like your TTS works wrong with HA voice proxy. Sorry, I'm no help here... Too specific. Try HA Discord, they might have clues.