formatBCE/Respeaker-Lite-ESPHome-integration #57: TTS Connect timed out before data was ready

walloutlet, 2025-06-19:
Good day, apologies if creating this as an issue is the wrong process.  It was the only option and I'm not sure the best way to ask a question for this.  The  question being, what would cause the error message seen in the ESPHome device log snippet below?  

Error message (bigger log snippet below); 

[ann_read]: I (1338686) esp-x509-crt-bundle: Certificate validated
[ann_read]: W (1343957) HTTP_CLIENT: Connection timed out before data was ready!
[speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[speaker_media_player:326]: The announcement pipeline's file reader encountered an error.

Below the log snippet is what the full response should be but I can see in the ESPHome log that it is only accepting a certain amount of text.  This only seems to occur when there is a longer TTS.  Otherwise everything works perfectly fine.

I am using the ReSpeaker Lite with the latest code (2025.6.0) and Home Assistant 2025.6.1.

Appreciate your insight.

Log Snippet:
```
[13:16:05][D][voice_assistant:758]: Starting STT by VAD
[13:16:05][D][light:036]: 'led_internal' Setting:
[13:16:05][D][light:051]:   Brightness: 100%
[13:16:05][D][light:058]:   Red: 100%, Green: 20%, Blue: 100%
[13:16:07][D][voice_assistant:598]: Event Type: 12
[13:16:07][D][voice_assistant:762]: STT by VAD end
[13:16:07][D][voice_assistant:456]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[13:16:07][D][voice_assistant:463]: Desired state set to AWAITING_RESPONSE
[13:16:07][D][voice_assistant:456]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[13:16:07][D][light:036]: 'led_internal' Setting:
[13:16:07][D][light:051]:   Brightness: 100%
[13:16:07][D][light:058]:   Red: 100%, Green: 20%, Blue: 100%
[13:16:07][D][light:109]:   Effect: 'Fast Pulse'
[13:16:07][D][voice_assistant:456]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[13:16:10][D][voice_assistant:598]: Event Type: 4
[13:16:10][D][voice_assistant:626]: Speech recognised as: "What's my daily summary?"
[13:16:10][D][voice_assistant:598]: Event Type: 5
[13:16:10][D][voice_assistant:631]: Intent started
[13:16:13][D][voice_assistant:598]: Event Type: 6
[13:16:13][D][voice_assistant:598]: Event Type: 7
[13:16:13][D][voice_assistant:656]: Response: "As of today, June 19, the weather is mostly sunny, windy and not as hot with a real feel temperature of 98.1 degrees, it is recommended to dress in light and breathable clothing to stay cool, also consider wearing sunglasses and a hat to protect yourself from the sun, and don't forget to apply sunscreen with a high UV index of 10. Looking ahead to the next 48 hours, tomorrow's weather forecast is partly sunny and very warm with a real feel temperature of 93.6 deg
[13:16:13][D][light:036]: 'led_internal' Setting:
[13:16:13][D][light:051]:   Brightness: 100%
[13:16:13][D][light:058]:   Red: 20%, Green: 100%, Blue: 100%
[13:16:13][D][light:109]:   Effect: 'Slow Pulse'
[13:16:13][D][voice_assistant:598]: Event Type: 8
[13:16:13][D][voice_assistant:678]: Response URL: "https://haos.url/api/tts_proxy/iAOLF0mzr00NiHoOKer5Dw.flac"
[13:16:13][D][voice_assistant:456]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[13:16:13][D][voice_assistant:463]: Desired state set to STREAMING_RESPONSE
[13:16:13][D][media_player:074]: 'respeaker-satellite-bedroom' - Setting
[13:16:13][D][media_player:081]:   Media URL: https://haos.url/api/tts_proxy/iAOLF0mzr00NiHoOKer5Dw.flac
[13:16:13][D][media_player:087]:  Announcement: yes
[13:16:13][D][speaker_media_player:406]: State changed to ANNOUNCING
[13:16:13][D][voice_assistant:598]: Event Type: 2
[13:16:13][D][voice_assistant:697]: Assist Pipeline ended
[13:16:14][D][esp-idf:000][ann_read]: I (1338686) esp-x509-crt-bundle: Certificate validated
[13:16:19][D][esp-idf:000][ann_read]: W (1343957) HTTP_CLIENT: Connection timed out before data was ready!
[13:16:19][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[13:16:19][E][speaker_media_player:326]: The announcement pipeline's file reader encountered an error.
[13:16:21][D][esp-idf:000][ann_read]: I (1345108) esp-x509-crt-bundle: Certificate validated
[13:16:26][D][esp-idf:000][ann_read]: W (1350382) HTTP_CLIENT: Connection timed out before data was ready!
[13:16:26][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[13:16:26][E][speaker_media_player:326]: The announcement pipeline's file reader encountered an error.
[13:16:26][D][speaker_media_player:406]: State changed to IDLE
[13:16:26][D][voice_assistant:329]: Announcement finished playing
[13:16:26][D][voice_assistant:456]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[13:16:26][D][voice_assistant:463]: Desired state set to RESPONSE_FINISHED
[13:16:26][D][voice_assistant:456]: State changed from RESPONSE_FINISHED to IDLE
[13:16:26][D][voice_assistant:463]: Desired state set to IDLE
[13:16:26][D][light:036]: 'led_internal' Setting:
[13:16:26][D][light:047]:   State: OFF
[13:16:26][D][light:109]:   Effect: 'None'
[13:17:00][D][text_sensor:069]: 'Current device time': Sending state '17:17'
```

The response should read:
`As of today, June 19, the weather is mostly sunny, windy and not as hot with a real feel temperature of 98.1 degrees, it is recommended to dress in light and breathable clothing to stay cool, also consider wearing sunglasses and a hat to protect yourself from the sun, and don't forget to apply sunscreen with a high UV index of 10. Looking ahead to the next 48 hours, tomorrow's weather forecast is partly sunny and very warm with a real feel temperature of 93.6 degrees, so it's still important to wear light and breathable clothing, and consider bringing an umbrella as there is a chance of rain, also make sure to stay hydrated by drinking plenty of water. In terms of calendar events, today is Juneteenth, a United States holiday, so some businesses and institutions may be closed, on Friday, June 20, it's pay day for both Brandon and Colby, Brandon, there are no other upcoming events or holidays for the next 48 hours, except for National Indigenous Peoples Day in Canada on Saturday, June 21, which may not be relevant to Brandon and Colby's schedules.`

formatBCE, 2025-06-19:
Hey!
There's a lot of things that can add to this issue.
Here are some thoughts:

1. HTTPS is usually having troubles with voice pipeline, and at the minimum - adds to delays. Try setting internal HTTP URL to your HA server in network settings.
2. What are you using for TTS? I think HA voice proxy just isn't getting the response from TTS in time (or can't convert the audio in time), so pipeline is stopping on timeout.
3. Try playing that file by pasting URL to the browser. Will it play? Will it choke?
4. All this is actually limitation of esp32s3, networking and timeouts. I saw many of same requests in HA and ESPHome Discord servers. Unfortunately, looks like we have to live with it so far. There are changes coming to the way speech is generated and processed (streaming responses), so we should get there eventually.

closed 2025-06-19 by formatBCE

walloutlet, 2025-06-19:
So a couple items with this, which seems to have gotten it to work.  

After digging through the bowels of Google a little bit I ran across this Home Assistant Community thread --> https://community.home-assistant.io/t/http-timeout-for-voice-assistant-pe-even-though-the-response-is-recieved/834200

The thread points to a timeout value defined in audio_reader.cpp, which is set to 5000 ms by default.   I adjusted this setting from 5000ms to 15000 ms.  The file I modified was the 'src/esphome/components/audio/audio_reader.cpp'. 

static const uint32_t CONNECTION_TIMEOUT_MS = 5000;

to

static const uint32_t CONNECTION_TIMEOUT_MS = 15000;

However, that didn't seem to work but did produce a slightly different output in logs and action.

Throwing a hail marry, one of the things I noticed is that the long response given by the LLM was broken up into paragraphs.  After adding "Your response should continue no emojis, new lines or bullets." to the prompt the LLM response was a single paragraph with no line breaks.

Surprisingly, this seems to have made it work.  Still need the timeout setting > 5000 ms but also need to make sure there is no new lines in the response.  That sounds silly to me and could be a Piper handling issue but I can get a consistent long response now to playback.

Hope that helps someone else, perhaps there is a way to add a configurable timeout value in Home Assistant or the source code that can be set to make this easier to adjust?