formatBCE/Respeaker-Lite-ESPHome-integration #58: Failed config when trying to Install

Scooby7274, 2025-07-07:
Not sure what I am doing wrong ,but I have done the flashing part,but I'm having issues with the yaml.

I created a **"common"** folder under the Esphome. Created a file called **"respeaker-satellite-base.yaml"**

I then go to Esphome and use the **"respeaker-satellite-dashboard-example.yaml"** and change the wifi ssd and password to match my network.

When I try to install I get this error..

**INFO ESPHome 2025.6.3
INFO Reading configuration /config/esphome/respeaker.yaml...
Failed config
esphome: [source /config/esphome/respeaker.yaml:4]
  Platform missing. You must include one of the available platform keys: rp2040, host, esp32, libretiny, rtl87xx, esp8266, bk72xx.
  name: respeaker-satellite
  friendly_name: respeaker-satellite**

Scooby7274, 2025-07-07:
Will the errors be in Red? I will have to run it again. Thanks

Scooby7274, 2025-07-07:
INFO ESPHome 2025.6.3
INFO Reading configuration /config/esphome/respeaker.yaml...
INFO Updating https://github.com/formatBCE/esphome@respeaker_microphone
INFO Updating https://github.com/formatBCE/Respeaker-Lite-ESPHome-integration@main
INFO Detected timezone 'America/New_York'
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

script: [source /config/esphome/common/respeaker-satellite-base.yaml:628]
  - id: control_leds
    then: 
      - lambda: |-
          id(check_if_timers_active).execute();
          if (id(is_timer_active)){
            id(fetch_first_active_timer).execute();
          }
          if (id(improv_ble_in_progress)) {
            id(control_leds_improv_ble_state).execute();
          } else if (id(init_in_progress)) {
            id(control_leds_init_state).execute();
          } else if (!id(wifi_id).is_connected() || !id(api_id).is_connected()){
            id(control_leds_no_ha_connection_state).execute();
          } else if (id(user_button).state) {
            id(control_leds_center_button_touched).execute();
          } else if (id(timer_ringing).state) {
            id(control_leds_timer_ringing).execute();
          } else if (id(voice_assistant_phase) == 2) {
            id(control_leds_voice_assistant_waiting_for_command_phase).execute();
          } else if (id(voice_assistant_phase) == 3) {
            id(control_leds_voice_assistant_listening_for_command_phase).execute();
          } else if (id(voice_assistant_phase) == 4) {
            id(control_leds_voice_assistant_thinking_phase).execute();
          } else if (id(voice_assistant_phase) == 5) {
            id(control_leds_voice_assistant_replying_phase).execute();
          } else if (id(voice_assistant_phase) == 11) {
            id(control_leds_voice_assistant_error_phase).execute();
          } else if (id(voice_assistant_phase) == 10) {
            id(control_leds_voice_assistant_not_ready_phase).execute();
          } else if (id(is_timer_active)) {
            id(control_leds_timer_ticking).execute();
          } else if (id(alarm_on).state && !id(led_respeaker_onboard).remote_values.is_on()) {
            id(control_leds_alarm_active).execute();
          } else if (id(voice_assistant_phase) == 1) {
            id(control_leds_voice_assistant_idle_phase).execute();
          }
  - id: control_leds_improv_ble_state
    then: 
      - light.turn_on: 
          
          invalid number.
          brightness: return max( id(led_respeaker_onboard).current_values.get_brightness() , 0.2f );
          red: 100%
          green: 89%
          blue: 71%
          id: led_internal
          effect: Slow Pulse
  - id: control_leds_init_state
    then: 
      - if: 
          condition: 
            wifi.connected: 
          then: 
            - light.turn_on: 
                brightness: return max( id(led_respeaker_onboard).current_values.get_brightness() , 0.2f );
                red: 9%
                green: 73%
                blue: 95%
                id: led_internal
[... 232 lines cut ...]
    then: 
      - homeassistant.event: 
          event: esphome.alarm_ringing
  - id: send_tts_uri_event
    parameters: 
      tts_uri: string
    then: 
      - homeassistant.event: 
          event: esphome.tts_uri
          data: 
            uri: return tts_uri;
  - id: send_stt_text_event
    parameters: 
      stt_text: string
    then: 
      - homeassistant.event: 
          event: esphome.stt_text
          data: 
            text: return stt_text;
  - id: publish_current_time
    then: 
      - lambda: |-
          // Publish current time
          auto time_now = id(homeassistant_time).now();
          id(current_time).publish_state(time_now.strftime("%H:%M"));

Scooby7274, 2025-07-07:
When I copied and pasted the yaml file,I got this error...

**unknown tag !<!lambda> (253:69)

 250 |  ... 
 251 |  ... 
 252 |  ... 
 253 |  ... turn id(mute_switch_on_sound);
------------------------------------------^
 254 |  ...** 

I Did run it through a yaml fixer and use that and it said it was ok. Could that be the issue.

Scooby7274, 2025-07-07:
Thats what I copied and pasted and got teh errors mentioned above.

Scooby7274, 2025-07-07:
Should I try to reflash the firmware?

formatBCE, 2025-07-07:
> Should I try to reflash the firmware? 

To re-flash firmware,  you need working config. :)
Also, re-flashing the device won't help with ESPHome issue.

Make sure you have latest ESPHome, and double check all files you copied.

Scooby7274, 2025-07-07:
I'm not sure,I'll have to come back to it later.

formatBCE, 2025-07-07:
There's no `lambda` part there. It's required. 

PLEASE make sure you copied and pasted common config without any changes. Don't use some YAML fixers or other things. Just select all - copy, then select all - delete - paste.

formatBCE, 2025-07-07:
Glad to hear!
I always use Studio Code add-on, it's the best. :)

closed 2025-07-07 by formatBCE

Scooby7274, 2025-07-07:
I have that installed,not sure why I didn't use it.  It's still compiling,so I should be good now. Tahnsk for the Help,appreciate it.